TrustedFirmware-M/trusted-firmware-m #6: 04: Stack overflow in `secure_fw/spm/core/spm_local_connection.c#spm_allocate_connection`

hunkob, 2024-07-16:
# 04: Stack overflow in `secure_fw/spm/core/spm_local_connection.c#spm_allocate_connection`

## Summary

When handling incoming mailbox messages in `tfm_spm_agent_psa_call`, connection objects are dynamically allocated. If the local connection allocator is used (which is enabled in default builds of the SPE), connection objects are allocated from the stack. However, no checks are performed on the maximum number of allocated connections, and connections are not freed upon errors in `tfm_spm_client_psa_call`. This leads to an overflow of the available stack memory, corrupting the running process stack. The allocator implementation in `secure_fw/spm/core/spm_local_connection.c` seems to be missing vital checks and may require further review.

- **Impact:** Corruption of stack memory (RCE)
- **Severity:** Critical

## Technical Description

Partition calls use connections that are allocated within `tfm_spm_agent_psa_call`:

- [`tfm_spm_client_psa_call`](https://git.trustedfirmware.org/TF-M/trusted-firmware-m.git/+/refs/tags/TF-Mv2.0.0/secure_fw/spm/core/psa_call_api.c#154)
  - handles incoming calls and gets or creates corresponding connections with `spm_get_connection`
  - does not clean up connections on error due to not calling the message handler and `IS_STATIC_HANDLE(handle)`](https://git.trustedfirmware.org/TF-M/trusted-firmware-m.git/+/refs/tags/TF-Mv2.0.0/secure_fw/spm/core/psa_call_api.c#173) being `false`
- [`spm_get_connection`](https://git.trustedfirmware.org/TF-M/trusted-firmware-m.git/+/refs/tags/TF-Mv2.0.0/secure_fw/spm/core/spm_ipc.c#181)
  - allocations new (temporary) connections with `spm_allocate_connection`:
    - [`connection = spm_allocate_connection();`](https://git.trustedfirmware.org/TF-M/trusted-firmware-m.git/+/refs/tags/TF-Mv2.0.0/secure_fw/spm/core/spm_ipc.c#224)
- [`spm_allocate_connection`](https://git.trustedfirmware.org/TF-M/trusted-firmware-m.git/+/refs/tags/TF-Mv2.0.0/secure_fw/spm/core/spm_local_connection.c#82)
  - wrapper for `alloc_conn_from_stack_top`
- [`alloc_conn_from_stack_top`](https://git.trustedfirmware.org/TF-M/trusted-firmware-m.git/+/refs/tags/TF-Mv2.0.0/secure_fw/spm/core/spm_local_connection.c#24)
  - no stack overflow check
  - => can eventually overwrite whole stack
  - => this includes the process stack with saved return addresses => leading to `PC` control (RCE: remote code execution)

## Proposed fix

Complete refactor of allocator in `spm_local_connection.c`:

- Stack bounds should be checked before allocation (limit maximum of concurrent allocations)
- Free should deallocated the passed allocation (not the most recent allocation)
- Connection handle should correctly translate to the corresponding allocation (not the most recent allocation)

Extensive review and testing of the allocator should be done after implementation of the required fixes.

**Note:** Attached patch file is a *work-around* used during analysis of other issues and should not be used without further extensive review.

---

Related Notes / Potentially Non-Security Bugs:

- Connections should be freed after an error occurred
  - This is potentially due to connection handle not being detected as static handle ([`IS_STATIC_HANDLE(handle) == false`](https://git.trustedfirmware.org/TF-M/trusted-firmware-m.git/+/refs/tags/TF-Mv2.0.0/secure_fw/spm/core/psa_call_api.c#173))

## Reproducer

For each call that results in an error a new connection is allocated, eventually overflowing the stack with the new connection into the process stack.

Excerpt from `crash.log`:

```
mailbox_dispatch: CALL handle = 67990bca, type = 00000000, in_vec = 08031000, in_len = 3, out_vec = 08032000, out_len = 1
mailbox_direct_reply: result = ffffff7f (ERROR_PROGRAMMER_ERROR)
[...]
mailbox_dispatch: CALL handle = 5d8f01c0, type = 00000000, in_vec = 08031000, in_len = 3, out_vec = 08032000, out_len = 1
*stack corruption happens* (pointer on stack is overwritten)
Instruction access violation at d66cb1e4
Result: Crash { pc: d66cb1e4, ra: 100077ab, exception: 3 (HardFault) }
```

`100077ab` corresponds to [`secure_fw/spm/core/backend_sfn.c:57`](https://git.trustedfirmware.org/TF-M/trusted-firmware-m.git/+/refs/tags/TF-Mv2.0.0/secure_fw/spm/core/backend_sfn.c#57):

```c
((sfn_init_fn_t)p_target->p_ldinf->entry)(NULL);
```

**Note:** We omitted a detailed analysis of the stack corruption here for the sake of simplicity.

## Attachments

- [crash.log](https://github.com/user-attachments/files/16248146/crash.log): log file of crashing input (including sent messages)
- [frametrace-1012527.log](https://github.com/user-attachments/files/16248147/frametrace-1012527.log): trace of all called function + context of crash
- [0001-work-around-for-stack-overflow-in-connection-less-pa.patch](https://github.com/user-attachments/files/16248145/0001-work-around-for-stack-overflow-in-connection-less-pa.patch): *work-around* patch file, **not intended as fix**

nicola-mazzucato-arm, 2024-09-26:
Hi @hunkob , sorry my slow reply.

By looking at the code it seems that the following patch could be a valid fix for this issue: https://review.trustedfirmware.org/c/TF-M/trusted-firmware-m/+/31350

Would you be able to cherry-pick and re-test?
Many thanks in advance

adeaarm, 2024-09-27:
release/2.1.x cherry-pick: https://review.trustedfirmware.org/c/TF-M/trusted-firmware-m/+/31300