TrustedFirmware-M/trusted-firmware-m #11: Misplaced #endifs cause -Werror=implicit-function-declaration and -Werror=switch-unreachable in cc3xx_aes.c

Einhornhool, 2024-10-04:
## Summary
Hi!
I noticed this issue in the low level cc3xx driver.
Three misplaced #endifs cause compile time errors when CCM is not enabled. 

https://github.com/TrustedFirmware-M/trusted-firmware-m/blob/7684b44dbe1bc21a3abf8c9dc626776db50fbb89/platform/ext/target/arm/drivers/cc3xx/low_level_driver/src/cc3xx_aes.c#L735

```
#if defined(CC3XX_CONFIG_AES_CCM_ENABLE)
    case CC3XX_AES_MODE_CCM:
#endif /* defined(CC3XX_CONFIG_AES_CCM_ENABLE) */
        cc3xx_lowlevel_set_engine(CC3XX_ENGINE_AES);
        break;
    default:
        return;
```

https://github.com/TrustedFirmware-M/trusted-firmware-m/blob/7684b44dbe1bc21a3abf8c9dc626776db50fbb89/platform/ext/target/arm/drivers/cc3xx/low_level_driver/src/cc3xx_aes.c#L827
```
#ifdef CC3XX_CONFIG_AES_CCM_ENABLE
    case CC3XX_AES_MODE_CCM:
#endif /* CC3XX_CONFIG_AES_CCM_ENABLE */
#ifdef CC3XX_CONFIG_AES_TUNNELLING_ENABLE
        set_mode(CC3XX_AES_MODE_CTR);
        set_tun1_mode(CC3XX_AES_MODE_CBC_MAC);
        if (aes_state.direction == CC3XX_AES_DIRECTION_ENCRYPT) {
            /* Set AES_TUN_B1_USES_PADDED_DATA_IN, AES_TUNNEL0_ENCRYPT and
             * AES_OUTPUT_MID_TUNNEL_DATA */
            P_CC3XX->aes.aes_control |= 0b111 << 23;
        } else {
            /* Set AES_OUTPUT_MID_TUNNEL_DATA and AES_TUNNEL_B1_PAD_EN */
            P_CC3XX->aes.aes_control |= 0b11 << 25;
            /* Unset AES_TUNNEL0_ENCRYPT */
            P_CC3XX->aes.aes_control &= ~(0b1 << 24);
        }

        /* Set AES_TUNNEL_IS_ON */
        P_CC3XX->aes.aes_control |= 0B1U << 10;

        cc3xx_lowlevel_set_engine(CC3XX_ENGINE_AES);
#else
        /* Withut tunnelling, we just perform CBC_MAC */
        *write_output = false;
        return;
#endif /* CC3XX_CONFIG_AES_TUNNELLING_ENABLE */
        break;
    default:
        cc3xx_lowlevel_set_engine(CC3XX_ENGINE_AES);
```

https://github.com/TrustedFirmware-M/trusted-firmware-m/blob/7684b44dbe1bc21a3abf8c9dc626776db50fbb89/platform/ext/target/arm/drivers/cc3xx/low_level_driver/src/cc3xx_aes.c#L1131

```
#ifdef CC3XX_CONFIG_AES_CCM_ENABLE
    case CC3XX_AES_MODE_CCM:
#endif /* CC3XX_CONFIG_AES_CCM_ENABLE */
        err = ccm_finish(tag);
        break;
    default:
        break;
```

## Proposed Fix:
```
#if defined(CC3XX_CONFIG_AES_CCM_ENABLE)
    case CC3XX_AES_MODE_CCM:
        cc3xx_lowlevel_set_engine(CC3XX_ENGINE_AES);
        break;
#endif /* defined(CC3XX_CONFIG_AES_CCM_ENABLE) */
    default:
        return;
```

```
#ifdef CC3XX_CONFIG_AES_CCM_ENABLE
    case CC3XX_AES_MODE_CCM:
#ifdef CC3XX_CONFIG_AES_TUNNELLING_ENABLE
        set_mode(CC3XX_AES_MODE_CTR);
        set_tun1_mode(CC3XX_AES_MODE_CBC_MAC);
        if (aes_state.direction == CC3XX_AES_DIRECTION_ENCRYPT) {
            /* Set AES_TUN_B1_USES_PADDED_DATA_IN, AES_TUNNEL0_ENCRYPT and
             * AES_OUTPUT_MID_TUNNEL_DATA */
            P_CC3XX->aes.aes_control |= 0b111 << 23;
        } else {
            /* Set AES_OUTPUT_MID_TUNNEL_DATA and AES_TUNNEL_B1_PAD_EN */
            P_CC3XX->aes.aes_control |= 0b11 << 25;
            /* Unset AES_TUNNEL0_ENCRYPT */
            P_CC3XX->aes.aes_control &= ~(0b1 << 24);
        }

        /* Set AES_TUNNEL_IS_ON */
        P_CC3XX->aes.aes_control |= 0B1U << 10;

        cc3xx_lowlevel_set_engine(CC3XX_ENGINE_AES);
#else
        /* Withut tunnelling, we just perform CBC_MAC */
        *write_output = false;
        return;
#endif /* CC3XX_CONFIG_AES_TUNNELLING_ENABLE */
#endif /* CC3XX_CONFIG_AES_CCM_ENABLE */
        break;
    default:
        cc3xx_lowlevel_set_engine(CC3XX_ENGINE_AES);
```
```
#ifdef CC3XX_CONFIG_AES_CCM_ENABLE
    case CC3XX_AES_MODE_CCM:
        err = ccm_finish(tag);
        break;
#endif /* CC3XX_CONFIG_AES_CCM_ENABLE */
    default:
        break;
```

adeaarm, 2024-10-05:
Thanks for this. Will you open a patch for those? Otherwise I can pick up your proposed fix here, but I will have to use my own Signed-off-by: <> message if I push it myself.

Einhornhool, 2024-10-06:
> [Verfolgungsbild][https://github.com/notifications/beacon/AGGNOSPQDSP5EXVDQVHSNLLZ2A52XA5CNFSM6AAAAABPMH6QTSWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUOYNQFQ.gif]

Hi!
Sure, I'll try!

Einhornhool, 2024-10-08:
Here it is: https://review.trustedfirmware.org/c/TF-M/trusted-firmware-m/+/31681

adeaarm, 2024-10-08:
Won't go in `release/2.1.x` as this is just a fix for compiler warnings which break the build in case `-Werror` is set

closed 2024-10-08 by adeaarm