TrustedFirmware-M/trusted-firmware-m #15: cc3xx/low-level/pka: SRAM size not configurable

mguetschow, 2024-10-21:
The cc3xx PKA engine SRAM size is unconditionally set to 6 KiB:
https://github.com/TrustedFirmware-M/trusted-firmware-m/blob/041629a4b91c9dcacf7708c3d15e27a3280f27b9/platform/ext/target/arm/drivers/cc3xx/low_level_driver/src/cc3xx_pka.c#L26

However, at least according to the documentation on the CryptoCell 310 for the nRF52840 at https://docs.nordicsemi.com/bundle/ps_nrf52840/page/cryptocell.html#ariaid-title106, cc310 only has 4 KiB of SRAM. Therefore, the SRAM size should adapt depending on the used cc3xx version.

I'm not sure at what level this should be configurable, probably via `cc3xx_config.h`: should there be something along the lines of `CC3XX_CONFIG_VERSION` being either 310 and 312, and the SRAM size adapting to it, or rather an explicit `CC3XX_CONFIG_PKA_SRAM_SIZE`? I'm happy to provide a patch for it, as soon as I know which approach I should take.

adeaarm, 2024-10-23:
Hi, thanks for this. I would prefer not having references to CryptoCell revisions in the code, so my preference would be to have it configurable and introducing a parameter CC3XX_CONFIG_PKA_RAM_SIZE. If you can provide a patch in that direction it would be great

adeaarm, 2024-10-23:
Or you could use ARCH_IS_CC310 which is used similarly in other parts of the codebase IIRC. If ARCH_IS_CC310 then PKA_RAM_SIZE is 4KiB, otherwise use 6KiB default

adeaarm, 2024-10-23:
We discussed this a bit more, I guess it is fine to have a config option like CC3XX_CONFIG_CC310 which will act similarly as the old ARCH_IS_CC310. Then you can init the PKA size based on that config option to 4 KB, or use the 6KB default

adeaarm, 2024-10-23:
Let's go with what @RcColes proposes here: https://review.trustedfirmware.org/c/TF-M/trusted-firmware-m/+/32950/4/platform/ext/target/arm/drivers/cc3xx/low_level_driver/src/cc3xx_dma.c#124

Also, now that you're working on it, could you please add a check during cc3xx_init against the PIDR/CIDR registers value so that if this new option is set, we check that we are effectively running at runtime on a CC310?

Thanks a lot for the patches.

mguetschow, 2024-10-23:
> Let's go with what @RcColes proposes here: https://review.trustedfirmware.org/c/TF-M/trusted-firmware-m/+/32950/4/platform/ext/target/arm/drivers/cc3xx/low_level_driver/src/cc3xx_dma.c#124

Done :)

> Also, now that you're working on it, could you please add a check during cc3xx_init against the PIDR/CIDR registers value so that if this new option is set, we check that we are effectively running at runtime on a CC310?

I'd be happy to do so if you hint me to the place where I'd find the correct value for those registers, AFAICT it's not mentioned in the publicly available CRYPTOCELL documentation by Nordic: https://docs.nordicsemi.com/bundle/ps_nrf52840/page/cryptocell.html

adeaarm, 2024-10-23:
> I'd be happy to do so if you hint me to the place where I'd find the correct value for those registers, AFAICT it's not mentioned in the publicly available CRYPTOCELL documentation by Nordic: [...]

I don't have up to date info as well unfortunately. We could do it in reverse, as you have access to CC-310 HW, just dump the content of the `P_CC3XX->id.peripheral_id_0`, and we can assume that in your case that would identify the CC-310 revision

nicola-mazzucato-arm, 2024-10-24:
pick into release branch: https://review.trustedfirmware.org/c/TF-M/trusted-firmware-m/+/31733