TrustedFirmware-M/trusted-firmware-m #19: Services do not unmap IOVECS

brianq-ti, 2024-12-05:
If existing Crypto and Attestation services are used as connection-based services with MM_IOVECS, input vecs are not unmapped after usage which results in panic upon the next call processing.

adeaarm, 2024-12-05:
Thanks for raising this on the mailing list. The original design for the Crypto and Attestation service was for them to be stateless service, but I appreciate that different integrations might want to interface with the service in a stateful manner, and there is nothing in the service itself that should prevent this. So I will cherry pick the patches that fix this issue on mainline, once they have been merged.

nicola-mazzucato-arm, 2024-12-06:
Proposed fix: https://review.trustedfirmware.org/q/topic:%22mm-invec-unmap-crypto-att%22
From the email thread within the mailing list, @brianq-ti confirmed that the set provides a fix for this issue.

Thanks

nicola-mazzucato-arm, 2024-12-06:
There's an outstanding outcome w.r.t. to automatic unmapping that should be carried out by the framework, as stated in the FF-M-v1.1 spec, section 5.2.4:
`The framework unmaps an input or output vector either in response to an unmapping call from the RoT
Service, or automatically when message processing is completed.`
This second part of the statement is not implemented and shall be tracked in another github issue.

nicola-mazzucato-arm, 2024-12-19:
Fixed with this series: https://review.trustedfirmware.org/q/topic:%22mm-invec-unmap-crypto-att%22
Thanks

nicola-mazzucato-arm, 2025-01-03:
Added for review in release branch at: https://review.trustedfirmware.org/q/topic:%22mm-invec-unmap-crypto-att-release/2.1.x%22

brianq-ti, 2025-01-06:
> ZjQcmQRYFpfptBannerStart
> This message was sent from outside of Texas Instruments.
>
> Do not click links or open attachments unless you recognize the source of this email and know the content is safe.
>
>     Report Suspicious  <https://us-phishalarm-ewt.proofpoint.com/EWT/v1/G3vK!tHdmlV8nHuexpM1Md3tp9P_jFMPUDFkedOLs-Tt-tvAU8ZsrOgWZ34G3YrBhCt9yegtafGj44n4SAFRKzD8T4h2-TavhEvgJ4EFGwIaRUWdlNKf_R8T_QQI$>   ‌
>
> ZjQcmQRYFpfptBannerEnd

Looks good, thanks!

Regards,
Brian

adeaarm, 2025-02-13:
Closed via d36ecf3ab0933b89795b6cfb0552cf299466829a and 57360f620cc3c98c43de5f3905d1a3103c36c874

closed 2025-02-13 by adeaarm