TrustedFirmware-M/trusted-firmware-m #22: Clarification on boot seed

augustozanellato, 2025-02-07:
Hi,
I'm doing a PoC on initial attestation for an university project using a RP2350 board and I'm not really understanding why boot seed is stored in OTP, shouldn't it be a value changing on each boot?
Pico SDK conveniently exposes one via a ROM function (rom_get_boot_random).

Another similar doubt I have is w.r.t. implementation id, couldn't it be read from the board id (pico_get_unique_board_id)? Board id is factory programmed in otp so it should actually be unique.

adeaarm, 2025-02-07:
Hi Augusto, we normally use the Github issue tracker only to track issues affecting TF-M releases, hence it would be better if you repost your questions to tf-m@lists.trustedfirmware.org for best possible outreach and visibility. I am going now to close this issue but please repost on the mailing list.

Thank you!
/Antonio

closed 2025-02-07 by adeaarm

augustozanellato, 2025-02-07:
Hi Antonio,
Thanks for the response. I think i mixed stuff up and sent the email before signing up to the mailing list so it's pending moderator approval. Could you check for me please?
Thanks,
Augusto