TrustedFirmware-M/trusted-firmware-m #29: stm32u5 platform: crypto hw accelerator: SHSI clock is kept stopped after init causing crypto operations to timeout after wakup from stop

mkaranki, 2025-06-05:
In STM32 crypto accelerator driver code included in TF-M,  SHCI clock is enabled only in `crypto_hw_accelerator_init()` during crypto_init. As the clock is stopped after entering into stop modes, and not started automatically after that, any crypto operation using SAES crypto accelerator fails after wakup from stop mode. Impact is that crypto stops completely working if MCU low power modes are entered by non-secure execution environment.

SHSI clock is a clock used by SAES as per RM0456 STM document says:

[image]

RCC control register bit documentation states the clock gets disabled on stop mode:

[image]

I think that TF-M's STM32 crypto accelerator driver should enable SHSI clock when needed to allow the non-secure side to enter stop modes. Also keeping the clock unnecessarily running would be good to be avoided?

With the patch below, I can get the crypto accelerator to not jam after wakeup. That simply enables SHCI at the same points where SAES bus clock is enabled:

```diff
diff --git a/platform/ext/target/stm/common/hal/accelerator/aes_alt.c b/platform/ext/target/stm/common/hal/accelerator/aes_alt.c
index 2b279788d..a2052873c 100644
--- a/platform/ext/target/stm/common/hal/accelerator/aes_alt.c
+++ b/platform/ext/target/stm/common/hal/accelerator/aes_alt.c
@@ -139,6 +139,7 @@ static int aes_set_key(mbedtls_aes_context *ctx,
 
 #if defined(GENERATOR_HW_CRYPTO_DPA_SUPPORTED) && defined(HW_CRYPTO_DPA_AES)
     /* Enable SAES clock */
+    __HAL_RCC_SHSI_ENABLE();
     __HAL_RCC_SAES_CLK_ENABLE();
 #else
     /* Enable AES clock */
diff --git a/platform/ext/target/stm/common/hal/accelerator/gcm_alt.c b/platform/ext/target/stm/common/hal/accelerator/gcm_alt.c
index 9d6067354..765268048 100644
--- a/platform/ext/target/stm/common/hal/accelerator/gcm_alt.c
+++ b/platform/ext/target/stm/common/hal/accelerator/gcm_alt.c
@@ -101,6 +101,7 @@ void mbedtls_gcm_init( mbedtls_gcm_context *ctx )
 
 #if defined(GENERATOR_HW_CRYPTO_DPA_SUPPORTED) && defined(HW_CRYPTO_DPA_GCM)
    /* Enable SAES clock */
+    __HAL_RCC_SHSI_ENABLE();
     __HAL_RCC_SAES_CLK_ENABLE();
 #endif
     memset( ctx, 0, sizeof( mbedtls_gcm_context ) );
@@ -257,6 +258,7 @@ int mbedtls_gcm_setkey( mbedtls_gcm_context *ctx,
     }
 #if defined(GENERATOR_HW_CRYPTO_DPA_SUPPORTED) && defined(HW_CRYPTO_DPA_GCM)
     /* Enable SAES clock */
+    __HAL_RCC_SHSI_ENABLE();
     __HAL_RCC_SAES_CLK_ENABLE();
 #else
     __HAL_RCC_AES_CLK_ENABLE();
```

ahmadstm, 2025-06-06:
link for the patch to fix this issue: https://review.trustedfirmware.org/c/TF-M/trusted-firmware-m/+/38951

adeaarm, 2025-06-17:
Fixed through b4f636eac270b07832ee0c50b7b719665103d6d4 on mainline. @ahmadstm do you need backport on the LTS branch? Otherwise I will just close the ticket.