TrustedFirmware-M/trusted-firmware-m #30: stm: flash: driver doesn't disable icache when erasing or writing flash memory

vai-rikar, 2025-06-06:
At least with STM32U5, erasing or writing flash memory while ICACHE is enabled, causes ERRF bit to be set in ICACHE status register. From RM0456:

[image]

Here's a patch which disables  ICACHE before erase/write operations and re-enables it afterwards:
```
diff --git a/platform/ext/target/stm/common/hal/CMSIS_Driver/low_level_flash.c b/platform/ext/target/stm/common/hal/CMSIS_Driver/low_level_flash.c
index e0ef15138..139722b18 100644
--- a/platform/ext/target/stm/common/hal/CMSIS_Driver/low_level_flash.c
+++ b/platform/ext/target/stm/common/hal/CMSIS_Driver/low_level_flash.c
@@ -22,6 +22,7 @@
 #include "stm32hal.h"
 #include <stdio.h>
 #include "board.h"
+#include "boot_hal_cfg.h"
 
 #ifndef ARG_UNUSED
 #define ARG_UNUSED(arg)  ((void)arg)
@@ -414,6 +415,83 @@ static int32_t Flash_ReadData(uint32_t addr, void *data, uint32_t cnt)
   }
 }
 
+#ifdef TFM_ICACHE_ENABLE
+#define ICACHE_DISABLE_TIMEOUT_VALUE           1U   /* 1ms */
+#define ICACHE_INVALIDATE_TIMEOUT_VALUE        1U   /* 1ms */
+
+static int stm32_icache_disable(void)
+{
+  int status = 0;
+  uint32_t tickstart;
+
+  /* Clear BSYENDF flag first and then disable the instruction cache
+   * that starts a cache invalidation procedure
+   */
+  CLEAR_BIT(ICACHE->FCR, ICACHE_FCR_CBSYENDF);
+
+  HAL_ICACHE_Disable();
+
+  /* Get tick */
+  tickstart = HAL_GetTick();
+
+  /* Wait for instruction cache to get disabled */
+  while (HAL_ICACHE_IsEnabled()) {
+    if ((HAL_GetTick() - tickstart) >
+            ICACHE_DISABLE_TIMEOUT_VALUE) {
+      /* New check to avoid false timeout detection in case
+       * of preemption.
+       */
+      if (HAL_ICACHE_IsEnabled()) {
+        status = ARM_DRIVER_ERROR_TIMEOUT;
+        break;
+      }
+    }
+  }
+
+  return status;
+}
+
+static void stm32_icache_enable(void)
+{
+  HAL_ICACHE_Enable();
+}
+
+static int icache_wait_for_invalidate_complete(void)
+{
+  int status = ARM_DRIVER_ERROR;
+  uint32_t tickstart;
+
+  /* Check if ongoing invalidation operation */
+  if (__HAL_ICACHE_GET_FLAG(ICACHE_FLAG_BUSY)) {
+    /* Get tick */
+    tickstart = HAL_GetTick();
+
+    /* Wait for end of cache invalidation */
+    while (!__HAL_ICACHE_GET_FLAG(ICACHE_FLAG_BUSYEND)) {
+      if ((HAL_GetTick() - tickstart) >
+          ICACHE_INVALIDATE_TIMEOUT_VALUE) {
+        break;
+      }
+    }
+  }
+
+  /* Clear any pending flags */
+  if (__HAL_ICACHE_GET_FLAG(ICACHE_FLAG_BUSYEND)) {
+    __HAL_ICACHE_CLEAR_FLAG(ICACHE_FLAG_BUSYEND);
+    status = 0;
+  } else {
+    status = ARM_DRIVER_ERROR_TIMEOUT;
+  }
+
+  if (__HAL_ICACHE_GET_FLAG(ICACHE_FLAG_ERROR)) {
+    __HAL_ICACHE_CLEAR_FLAG(ICACHE_FLAG_ERROR);
+    status = ARM_DRIVER_ERROR;
+  }
+
+  return status;
+}
+#endif // TFM_ICACHE_ENABLE
+
 static int32_t Flash_ProgramData(uint32_t addr,
                                  const void *data, uint32_t cnt)
 {
@@ -476,6 +554,15 @@ static int32_t Flash_ProgramData(uint32_t addr,
     return ARM_DRIVER_ERROR_PARAMETER;
   }
 
+#ifdef TFM_ICACHE_ENABLE
+  /* Disable icache, this will start the invalidation procedure.
+    * All changes(erase/write) to flash memory should happen when
+    * i-cache is disabled. A write to flash performed without
+    * disabling i-cache will set ERRF error flag in SR register.
+    */
+  stm32_icache_disable();
+#endif /* TFM_ICACHE_ENABLE */
+
   HAL_FLASH_Unlock();
   ARM_FLASH0_STATUS.busy = DRIVER_STATUS_BUSY;
   do
@@ -505,6 +592,16 @@ static int32_t Flash_ProgramData(uint32_t addr,
 
   ARM_FLASH0_STATUS.busy = DRIVER_STATUS_IDLE;
   HAL_FLASH_Lock();
+
+#ifdef TFM_ICACHE_ENABLE
+  icache_wait_for_invalidate_complete();
+
+  /* I-cache should be enabled only after the
+    * invalidation is complete.
+    */
+  stm32_icache_enable();
+#endif /* TFM_ICACHE_ENABLE */
+
   /* compare data written */
 #ifdef CHECK_WRITE
   if ((err == HAL_OK) && memcmp(dest, data, cnt))
@@ -587,12 +684,32 @@ static int32_t Flash_EraseSector(uint32_t addr)
 #else
     EraseInit.Page = page_number(&ARM_FLASH0_DEV, addr);
 #endif
+
+#ifdef TFM_ICACHE_ENABLE
+  /* Disable icache, this will start the invalidation procedure.
+    * All changes(erase/write) to flash memory should happen when
+    * i-cache is disabled. A write to flash performed without
+    * disabling i-cache will set ERRF error flag in SR register.
+    */
+  stm32_icache_disable();
+#endif /* TFM_ICACHE_ENABLE */
+
   ARM_FLASH0_STATUS.error = DRIVER_STATUS_NO_ERROR;
   HAL_FLASH_Unlock();
   ARM_FLASH0_STATUS.busy = DRIVER_STATUS_BUSY;
   err = HAL_FLASHEx_Erase(&EraseInit, &pageError);
   ARM_FLASH0_STATUS.busy = DRIVER_STATUS_IDLE;
   HAL_FLASH_Lock();
+
+#ifdef TFM_ICACHE_ENABLE
+  icache_wait_for_invalidate_complete();
+
+  /* I-cache should be enabled only after the
+    * invalidation is complete.
+    */
+  stm32_icache_enable();
+#endif /* TFM_ICACHE_ENABLE */
+
 #ifdef DEBUG_FLASH_ACCESS
   if (err != HAL_OK)
   {
diff --git a/platform/ext/target/stm/common/stm32u5xx/CMakeLists.txt b/platform/ext/target/stm/common/stm32u5xx/CMakeLists.txt
index f98d0d6a5..c1346f670 100644
--- a/platform/ext/target/stm/common/stm32u5xx/CMakeLists.txt
+++ b/platform/ext/target/stm/common/stm32u5xx/CMakeLists.txt
@@ -90,6 +90,7 @@ target_sources(platform_s
         ${CMAKE_CURRENT_SOURCE_DIR}/hal/Src/stm32u5xx_hal_cryp_ex.c
         ${CMAKE_CURRENT_SOURCE_DIR}/hal/Src/stm32u5xx_hal_i2c.c
         ${CMAKE_CURRENT_SOURCE_DIR}/hal/Src/stm32u5xx_hal_i2c_ex.c
+        ${CMAKE_CURRENT_SOURCE_DIR}/hal/Src/stm32u5xx_hal_icache.c
         ${STM_COMMON_DIR}/hal/Native_Driver/low_level_rng.c
         ${STM_COMMON_DIR}/hal/Native_Driver/nv_counters.c
 )
```

ahmadstm, 2025-06-06:
Thanks @vai-rikar for this. 
This folder is common with all ST platforms. So, I will test this with all another board.

ahmadstm, 2025-06-18:
Issue fixed : https://review.trustedfirmware.org/c/TF-M/trusted-firmware-m/+/39384

adeaarm, 2025-08-01:
Fixed through a68b8117235f12568bfd31e9c14b1bf0a1e7f957 on mainline

closed 2025-08-01 by adeaarm