TrustedFirmware-M/trusted-firmware-m #32: Copy/zero table block size alignment issue

gbrtth, 2025-06-16:
In `/platform/ext/common/gcc/tfm_isolation_s.ld.template` (line 246, 261)  the linker script’s copy/zero tables record region lengths in 32-bit words ( SIZEOF(...) / 4), but only align the start of each region—not the length. At start-up, the execution loops over whole words, so if a section’s size isn’t an exact multiple of four bytes, the final 1-3 bytes remain un-copied (or un-zeroed), leaving garbage data in those trailing bytes.

This issue may also affect other toolchains, but only GCC was investigated.

Anton-TF, 2025-07-22:
https://review.trustedfirmware.org/c/TF-M/trusted-firmware-m/+/40014
https://review.trustedfirmware.org/c/TF-M/trusted-firmware-m/+/40954

Anton-TF, 2025-07-22:
TF-M v2.1.2 is free from this bug being introduced later in ee0bebc648d3f2b6d983ee98d2b05e1568965e80